microbit-foundation/python-editor-v3 #207: Code structure highlighting: config options for experimentation

microbit-matt-hillsdon, 2021-07-22:
We've agreed to add a bunch of configuration options to make the code structure highlighting option more configurable.

We'll also add and/or document some CSS variables to enable easy experimenting with the colours used.

- [ ] As a first step we should collapse the separate implementations. The l-shaped-boxes one should have all the information needed to draw the others.
- [ ] Add functional options documented in Paper doc to settings
- [ ] Add colour options

microbit-matt-hillsdon, 2021-07-23:
This will probably also fix #204 or at least make it trivial.

microbit-matt-hillsdon, 2021-07-23:
Borders and l-shapes are a problematic combo if we stick with rounded corners

[image]

We might be able to make it work with an SVG but that would require a degree of reworking. Would probably let you round the inner corner of the L though.

microbit-matt-hillsdon, 2021-07-23:
Hard to say whether this left-edge-only boxes background combo is a good thing either:

[image]

microbit-matt-hillsdon, 2021-07-26:
We can't use CSS for hover in a straightforward way as L-shapes are made up of multiple boxes. I guess we might as well use event listeners for both cases for now, potentially simplifying in future.

microbit-matt-hillsdon, 2021-07-26:
I'll leave this on a branch for an initial round of feedback.

The most significant change for someone who doesn't engage with the options is stopping trimming trailing whitespace from the scope of a block. We could reinstate that but then I think it would be critical that we didn't highlight when we were in that whitespace as the mismatch is jarring.

jaustin, 2021-08-18:
Playing with the settings on the latest branch made https://review-python-editor-next.microbit.org/code-structure-settings-jonny/ I've realised a couple of things

The approach I took where I made the 'selected block' darker than the block background kinda limits (a) how strong the main block background colours can be and (b) means that we get into contrast issues sooner. For example it's harder to see the structure highlighting of the deeper nested bits of code once you've selected a lightly nested block...

I accidentally revisited the branch where Matt had made the selected block yellow, and I think there's a real advantage to the extra brightness you get from this.

Here are some examples of some options, the first is the existing proposal,  then a yellow, then a magenta option.
The yellow is a bit crunchy against the overall editor palette for me. The purple is better but I worry about contrast on some of the keyword highlighting.

[image]
[image]
[image]
[image]

microbit-matt-hillsdon, 2022-01-10:
All on :root (view from html element in dev tools):

--chakra-colors-code-blockBorder: rgba(123, 205, 194, 0.1);
--chakra-colors-code-blockBackground: rgba(123, 205, 194, 0.1);
--chakra-colors-code-blockBackgroundActive: lightyellow;
--chakra-colors-code-blockBorderActive: #262626;

closed 2022-07-13 by microbit-matt-hillsdon