microbit-foundation/python-editor-v3 #215: Pasting several lines into serial terminal causes "Bad status 8 -> 255" error and disconnect

pdg137, 2021-07-31:
## Bug Description

Pasting the following script into the terminal causes an instant disconnection:

```python
print("hello, world")
print("hello, world")
print("hello, world")
print("hello, world")
print("hello, world")
```
## Screenshots

[image]

## Environment

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrome
 - Latest: Version 92.0.4515.107 (Official Build) (64-bit)

## Additional context

Possibly related: #89.

I discovered this when writing long strings from Javascript using `device.serialWrite()`. My goal is to implement [microfs](https://github.com/ntoll/microfs) file downloads for getting data off of the micro:bit.

microbit-matt-hillsdon, 2021-08-02:
Reproduces for me on Mac.

I wanted to see if I could reproduce this on https://python.microbit.org/v/2 (as the serial code is very similar but the terminal is different) but there seems to be a short limit on the amount of pasted content there. It's not obviously something the v2 hterm setup is configuring though (Shift-Ins is paste there).

DETAILS.TXT from the micro:bit I reproduced with:

```
# DAPLink Firmware - see https://mbed.com/daplink
Unique ID: 9904360251974e450039900a00000041000000009796990b
HIC ID: 9796990b
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0255
Bootloader Version: 0254
Git SHA: 1436bdcc67029fdfc0ff03b73e12045bb6a9f272
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0xe5ec290c
Interface CRC: 0x5b5cc0f5
Remount count: 0
URL: https://microbit.org/device/?id=9904&v=0255
```

More investigation required.

microbit-matt-hillsdon, 2021-08-02:
Pasting 63 characters, e.g. `012345678901234567890123456789012345678901234567890123456789012` results in the disconnect. Pasting 62 characters is OK for me. It doesn't seem to be related to lines as such.

Looking more closely, you get the same error in https://python.microbit.org/v/2 but it seems to recover and the REPL still works afterwards. The pasted content isn't visible. The difference might just be that the error handling is more systematic in the alpha so we trigger a disconnect because WebUSB is in an error state. Still need to understand the cause.

microbit-matt-hillsdon, 2021-08-02:
Possible root cause in previous discussion: https://github.com/bbcmicrobit/PythonEditor/issues/280#issuecomment-546854603

microbit-matt-hillsdon, 2021-08-06:
Reopening for follow-up:

- [ ] Check DAPJs with node-usb transport. Raise DAPJs bug asking if this should be fixed in the WebUSB transport. See theory at https://github.com/microbit-foundation/python-editor-next/pull/218#discussion_r684238421
- [ ] Consider fix in V2 editor (raise bug etc.)

microbit-matt-hillsdon, 2021-08-06:
Confirmed fixed in the alpha deployment version  0.1.0-dev.732.

Darshnik253, 2022-02-09:
I am making project smart in that i have combined bluetooth and usb serial communication code. But if i run both simultaneously then its not working(getting error 020 sad icon). so could you help in that?

microbit-matt-hillsdon, 2022-02-09:
> I am making project smart in that i have combined bluetooth and usb serial communication code. But if i run both simultaneously then its not working(getting error 020 sad icon). so could you help in that?

Hi @Darshnik253, please get in touch with support via https://support.microbit.org. See [this support article](https://support.microbit.org/support/solutions/articles/19000097280-020) for details on the 020 error.

closed 2026-08-30 by microbit-matt-hillsdon