microbit-foundation/python-editor-v3 #217: In serial terminal, the keyboard shortcuts Ctrl+C and Ctrl+D are broken or confusing

pdg137, 2021-08-01:
## Bug Description

The following keyboard shortcuts have issues in the xterm / serial terminal / SerialArea:

* Even though Ctrl+V works for paste, Ctrl+C does not copy text.  There's also no option to copy in the right-click menu. Ctrl+C doesn't even copy the selected text if you click out of the xterm control first. So even though you can select, there appears to be no way to copy.
* Ctrl+D does a "soft reboot" but confusingly also creates a browser bookmark.

## How To Reproduce

Steps to reproduce the behavior:
1. Connect to a micro:bit.
2. Try Ctrl+C and Ctrl+D in the serial terminal area.

## Expected behavior

* An alternate key like Ctrl+Shift+C should be available for copy and mentioned in the introductory text along with Ctrl+C.
* There should be user-interface widgets such as buttons or a right-click menu to support copy and paste.
* The Ctrl+D shortcut should block the bookmark feature of Chrome.

## Screenshots

Can select text but no way to copy:

[image]

Annoying bookmark-and-soft-reboot shortcut:

[image]
## Environment

**Desktop (please complete the following information):**
 - OS: Windows 10
 - Chrome
 - Version 92.0.4515.107

microbit-matt-hillsdon, 2021-08-02:
Thanks for the great report. I'll look at this one later in the week.

Ctrl-C will need to remain unbound in order to interrupt a running micro:bit program. Ctrl-Shift-C/V is a common solution but I'll look at others and have a think about making this discoverable. https://python.microbit.org/v/2 copies text on selection and shows an icon indicating it has done so, which might be surprising for some but is at least very discoverable. That is default hterm behaviour (v2 uses hterm, the alpha uses xterm).

I've used Ctrl-D lots, but on Mac mostly, so I'd missed that it also creates a bookmark on Windows.

Useful reference: https://github.com/xtermjs/xterm.js/issues/2478

closed 2021-08-02 by microbit-matt-hillsdon

microbit-matt-hillsdon, 2021-08-02:
Fixed the triggering of the bookmark action in #219 and added Ctrl-Shift+C for copy in the same PR.

I've moved discussion of discoverability and/or documentation of these actions to #220 as I didn't want to block the bug fixes on agreeing UI changes which might not be a priority for a while.

microbit-matt-hillsdon, 2021-08-02:
Fix for this is in v0.1.0-dev.721 which is deployed at https://python.microbit.org/v/alpha

pdg137, 2021-08-03:
Tested it and those keys all seem good now, thanks for the quick fix!