microbit-foundation/python-editor-v3 #253: Full code structure highlight doesn't update after horizontal scroll

martinwork, 2021-08-25:
## Bug Description

Full code structure highlight doesn't update after horizontal scroll 

## How To Reproduce

Steps to reproduce the behavior:
1. Go to 'https://python.microbit.org/v/alpha'
2. Make the browser window narrower
3. Scroll across
4. See error

## Expected behavior

The shaded are should update

## Screenshots

[image]

## Environment

**Desktop (please complete the following information):**
 - OS: [e.g. iOS] Windows
 - Browser [e.g. chrome, safari] Chrome
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

## Additional context

Add any other context about the problem here.

JohnVidler, 2022-04-26:
Depth markers get weird with this too at deep depths:
[image]
[image]

microbit-matt-hillsdon, 2022-05-26:
The original issue has a work-in-progress fix on #752.

I expected it to fix https://github.com/microbit-foundation/python-editor-next/issues/253#issuecomment-1109649384 too but that one seems stranger. You can repro it with a narrow window and much less content. It happens to the first block that overflows the width of the window (and so requires scrolling).

microbit-matt-hillsdon, 2022-05-26:
The second issue on this ticket appears to be unrelated to code structure highlighting.

[image]

If you look carefully the line indicator also stops short. The content editable is not growing in this circumstance which seems likely to cause other bugs. The text still looks ok as it overflows. I'll see if I can reproduce it in a more minimal setup.

microbit-matt-hillsdon, 2022-05-26:
Raised https://github.com/microbit-foundation/python-editor-next/issues/753 for the editor size/scroll issue.

closed 2022-05-27 by microbit-matt-hillsdon