microbit-foundation/python-editor-v3 #263: [API] Prototype API documentation in sidebar

microbit-matt-hillsdon, 2021-09-14:
An initial prototype of API documentation in the sidebar that integrates with the editor.

This is early, exploratory work.

Aim to explore:
- Where we get the data from (stubs?, via Pyright?, precompute it for quick rendering?)
- Whether the stubs contain enough information to do a good job. Is manual curation required?
- What documentation improvements are necessary or desirable
- How we display it
- How we interact with the editor (e.g. drag and drop, other)
- Maybe, whether there's any interaction the other way, e.g. focus/goto docs as an alternative to docs on hover

microbit-matt-hillsdon, 2021-09-17:
Status comment (I'll edit this as we go):

-  [Branch with API documentation document-style](http://review-python-editor-next.microbit.org/apidocs)
-  [Branch with API documentation as an accordion](http://review-python-editor-next.microbit.org/apidocs-accordian) (yes, I misspelled the branch 😞 )

Both pretty rough! We could focus on design / layout questions, documentation raising quality, drag-and-drop, progressive disclosure of doc string detail... 

If we're generally happy with the direction, and don't seem likely to switch to manually curated docs for this API section, then there's work to be done to tidy up the Pyright side of this too.

### Which modules?

I’ve left out modules for now that aren’t in our readthedocs docs: antigravity, builtins, love, this, errno, array, collections, gc, struct, sys.

From discussions with Giles it seems like we might want some docs for the builtins. The stubs currently don't have docs but they exist somewhere...

antigravity and love should probably remain undocumented easter eggs (you can autocomplete to them). The others are probably fine to add but perhaps we'd want to make some modules less prominent as we add more? Though I think a basic/advanced split is harder to justify.

### Docstrings

I've truncated them at the moment but that's not really helpful. They're not suitable as-is.

Many don't contain a short summary that we could extract.

There is no systematic approach to parameter documentation.

They contain a lot of restructured text artifacts which we don't yet attempt to format or remove.

It'd be good to discuss what to do here.

### Types

I've shown type information. I think it's useful in general but I'm definitely not the target audience. The very few places that use generics look arcane (e.g. random.choose).

### More minor

- Dunder methods / operations (e.g. adding Images) aren't covered.
- Various issues raised against the stubs.

jaustin, 2021-09-23:
@microbit-matt-hillsdon this looks excellent, thanks! I'm really surprised by how good it looks and how *not* overwhelming it is even in this early form. Very promising.

## Which modules?

I think you've made a good first pass, and consider this ultimately an editorial choice. I think 'filesystem' probably wants to feature, a
I think the accordion doesn't work as well for me because it makes exploration hard, but OTOH the long lists of things like melodies and images slightly confuse the accordionless one. I wonder if there's a version of accordions that is 'auto entering' once one is opened, so you can keep on scrolling as if it's one big list, or contract it?  I( can't tell if in practice this will appear to be just the unexpanded list - I think it depends on when the collapse/re-expand happens, and whether you see the rest of the unexpanded items before )

Some simple things/responses to your questions:
 *  could we use 'returns' instead of '->': I think less notation is good, and I can stand the extra length of 'returns' I think. What do you think @microbit-giles 
 * ~As stated on Zoom, class inheritance isn't there, and I think it needs to be especially for buttons... https://microbit-micropython.readthedocs.io/en/v1.0.1/pin.html#classes~

### Docstrings

I think, as you say, this needs discussion. 

What you've generated looks good for giving us an overview and working out what needs to be done. I think you should probably just do as you suggest and create a good one-line summary for each item in the stubs. We can always refine these and it's good to have a starting point.

### Type information

I think on balance this is confusing. Partly because when is shown there is not the actual text you get in the editor, but without types it's way more similar.

 Places that are specifically confusing are things like this:
* `string: str | float | int` (💭 a string, that can also be a float or an int? ) 
* `choice(seq: Sequence[_T@choice]) -> _T@choice`  (💭 whoops, look like the micro:bit Foundation's docs parser broke and there's some mistake in the rendering here)
* `write(self: NeoPixel)` Not sure this is really the fault of the typings, actually.

But, the legitimate followup is do we just show parameters only with their name? The ReadTheDocs docs are more verbose, and are name first: (also note the disparity between our autogenerated name for the first parameter and the name in the docs, `string` versus `value`)
[image]

[image]

I'd be keen to have some kind of ‘expansion’ where you can see the more detailed info, but kinda think maybe type info could go into the docs

microbit-matt-hillsdon, 2021-11-02:
Raised https://github.com/microbit-foundation/python-editor-next/issues/301 for code pop-up
Raised https://github.com/microbit-foundation/python-editor-next/issues/302 for drop-down

Closing this initial prototype issue. Doubtless we'll raise more issues in future, e.g. to cover alignment with final designs and to address how we manage the content backing the micro:bit and Python toolkits, but this one isn't really actionable anymore.

closed 2021-11-02 by microbit-matt-hillsdon