microbit-foundation/python-editor-v3 #268: [Pyright] Fix _handleHighMemoryUsage for browser build and review other `process` use

microbit-matt-hillsdon, 2021-09-23:
The browser webpack build is compiling _handleHighMemoryUsage to:

```javascript
_handleMemoryHighUsage() {
    if (this._evaluator.getTypeCacheSize() > 75e4 || this._parsedFileCount > 1e3) {
        const e = Math.round({
            env: {},
            execArgv: [],
            cwd: ()=>"/"
        }.memoryUsage().heapUsed / 1048576);
        e > 1536 && (this._console.info(`Emptying type cache to avoid heap overflow. Heap size used: ${e}MB`),
        this._createNewEvaluator(),
        this._discardCachedParseResults(),
        this._parsedFileCount = 0)
    }
}
```

so it'll explode if we get over those limits. I suspect this might never happen for us, given our usecase, but it would be good to review this and other uses of process which has a very minimal DefinePlugin implementation.

microbit-matt-hillsdon, 2022-10-10:
Can we provide a useful implementation here? Failing that perhaps we just stub it out with a return 1 or similar.

microbit-robert, 2022-10-10:
Updated compilation:
```
_handleMemoryHighUsage() {
  if (
    this._evaluator.getTypeCacheSize() > 75e4 ||
    this._parsedFileCount > 1e3
  ) {
    const e = { env: {}, execArgv: [], cwd: () => "/" }.memoryUsage();
    if (e.heapUsed > 0.9 * e.rss) {
      const t = Math.round(e.rss / 1048576),
        n = Math.round(e.heapUsed / 1048576);
      this._console.info(
        `Emptying type cache to avoid heap overflow. Used ${n}MB out of ${t}MB`
      ),
        this._createNewEvaluator(),
        this._discardCachedParseResults(),
        (this._parsedFileCount = 0);
    }
  }
}
```

No other uses of `process` that need to be addressed, only `process.cwd`, `process.env.*`, etc.

PR #1038 (2022-10-19): Upgrade Pyright.
Closes #268
commit ed6b16bd (2022-10-19): Upgrade Pyright. (#1038)

Closes #268
closed 2022-10-19 by microbit-matt-hillsdon