microbit-foundation/python-editor-v3 #335: [Toolkits] DnD from Reference and Explore toolkits

microbit-matt-hillsdon, 2021-11-22:
Implement DnD from the Reference and Explore toolkits.

- On hover drag handles on the code snippets in Explore and the signature in Reference (tablet to be considered later, might require more changes).
- "Ghost code" in the editor, similar to previous spike.
- Drop behaviour as per the apidocs-dnd branch for reference drags, just insert and move caret to the end for code blocks.
- Show imports on drag? They're fixed regardless of where you drop could be clearer.

Desirable but later:
- Drop into the REPL. Does that affect showing imports on drag? Does it send Ctrl-C, then paste?

For discussion after:
- Insert into REPL button when connected via WebUSB. Perhaps "Run" or "Try it" ?

microbit-matt-hillsdon, 2021-12-09:
For the ghost code, I think we can make edits to the document annotating the transaction with the `Transaction.addToHistory.of(false)` annotation. 

This is done the collab module which has a similar problem of edits from two sources. So long as we take responsibility for undoing our changes we'll be fine. We're playing on easy mode compared to collab as we know no other changes can happen during the drag operation.

microbit-robert, 2021-12-15:
Possible future improvement:

- When dragging and dropping a code snippet into an area that should be a code block (i.e., underneath `while True:` or something similar), this code snippet should be indented to proper level, instead of falling back to the base line (no indentation).

commit b56ed048 (2022-01-12): Reference toolkit dnd (#403)

Similar style for draggable elements as in CM but without the pop-over behaviour that complicates Explore.

Drop experience that triggers signature help and positions the cursor between the brackets.
Various issues with the post-drop cursor position fixed that also affected Explore.

Closes #335

Raised #424 and #425 for future work.
closed 2022-01-12 by microbit-matt-hillsdon