microbit-foundation/python-editor-v3 #367: Add MakeCode-style "projects library"

duckida, 2021-12-12:
Hello! I was just trying out the new editor, and I love the features so far. The one thing that is is missing, though is the ability to store many projects on the editor, and open them through a library of projects. This will help teachers manage projects across multiple grades, and students work on many projects, at both home and school.

microbit-matt-hillsdon, 2021-12-13:
Thanks for your feedback.

We are prioritising an experience where users can get straight into coding without hitting a project management interface first. In case it isn't clear, the ‘hex’ file that you can download can be easily dragged back into the editor to reload the project. This also means teachers can use whichever existing system they already use for file storage, such as OneDrive, Google Drive, Dropbox or school LMS.

We appreciate that it’s a useful feature though and we might revisit once we’ve launched. This issue would be a great place for folks to add details of what they’re looking for in project storage. For example, local to your browser (with some risk of accidental loss, particularly in classroom setups) vs cloud storage (which)? Where would you use the feature? Classroom, code club, home, other? @duckida it would be great if you'd comment with these details and anything else you think would be helpful.

One option that will be available once the editor launches is micro:bit classroom. This doesn’t have local or cloud storage of projects, but it does enable teachers to store and resume whole-class sessions with a downloadable file. You can try this out with the current (V2) Python editor and MakeCode at https://classroom.microbit.org.

duckida, 2022-01-09:
I think that we should save to cloud(whichever one micro:bit foundation currently uses), so there is no accidental loss, as classroom computers often have buggy MDM systems. This would be useful in schools, where pupils will work on a program over-time, or at home, where a micro:bit enthusiast can store their previous programs