microbit-foundation/python-editor-v3 #387: GitHub integration

microbit-matt-hillsdon, 2021-12-21:
Issue to track requests for GitHub integration.

This is not a high priority issue for us while we're focused on the initial release of the editor.

If this is a feature that's of interest we'd love to understand a bit more about how you would use it. Please add a comment to the issue, or if you prefer, get in touch via the feedback link in the [alpha editor](https://python.microbit.org/v/alpha).

Some questions (but please do tell us anything you think might be useful):

1. How would you use the Python editor? For example, in the classroom, a code club, a library or home use. Tell us more about your micro:bit use.
2. Would students use GitHub integration themselves? Public or private repositories? GitHub classroom?
3. Are you focussed on teaching version control in particular or is GitHub used for version control in an activity that isn't really focussed on version control?

GingerGigiCat, 2022-02-01:
Sorry for the very delayed reply,

  1. I mainly use the new micropython editor for home use, but sometimes in a classroom. I often use add-ons/extensions such as the Kitronik View text:32, the Zip:64, and many breakout boards, so it would also be nice if there was an easy way to get the scripts for them instead of having to find them on the internet. (Not sure how easy this would be to implement though)
  2. Students would mainly use the integration in private repos as a way to easily access work and projects at home, and version control likely wouldn't be that necessary. GitHub classroom might be used on occasion, but only for giving students base files to work off of and possibly for submitting work.
  3. Teaching version control would not be focused on, GitHub would be primarily used for saving projects and potentially publishing publicly if students would like to.

GingerGigiCat, 2023-11-02:
@microbit-matt-hillsdon
Would this be looked into again, now that the editor has been released and is stable? Some kind of github integration would be really useful for version control, as well as for keeping a backup of code.
Thanks,
- Gigi