microbit-foundation/python-editor-v3 #402: Add Simulator feature

duckida, 2021-12-24:
I have found that the simulator in MakeCode can be inserted with an HTML `iframe`, such as the code below. The code to run on the micro:bit simulator is in MakeCode JavaScript, but it is possible we could create a Python version.

iFrame code:
`<iframe style="position:absolute;top:0;left:0;width:100%;height:100%;" allowfullscreen="allowfullscreen" frameborder="0" sandbox="allow-popups allow-forms allow-scripts allow-same-origin" src="/---run#nofooter=1" data-code="input.onButtonPressed(Button.A%2C%20()%20%3D%3E%20%7B%0A%20%20%20basic.showString(%22A%22)%3B%0A%7D)%3B%0Ainput.onButtonPressed(Button.B%2C%20()%20%3D%3E%20%7B%0A%20%20%20basic.showString(%22B%22)%3B%0A%7D)%3B%0Ainput.onPinPressed(TouchPin.P0%2C%20()%20%3D%3E%20%7B%0A%20%20%20basic.showString(%220%22)%3B%0A%7D)%3B%0Ainput.onPinPressed(TouchPin.P1%2C%20()%20%3D%3E%20%7B%0A%20%20%20basic.showString(%221%22)%3B%0A%7D)%3B%0Ainput.onPinPressed(TouchPin.P2%2C%20()%20%3D%3E%20%7B%0A%20%20%20basic.showString(%222%22)%3B%0A%7D)%3B%0Ainput.temperature()%0Ainput.compassHeading()%0Ainput.lightLevel()"></iframe>`

The code in MakeCode JS is after the `data-code` label.

microbit-matt-hillsdon, 2022-01-07:
Thanks @duckida. We do plan to work on a simulator in future but it's not our current focus.

To make a simulator with behaviour that matches MicroPython will require a significantly different approach to the MakeCode simulator.

For the moment, if you have a micro:bit device and a browser that supports WebUSB (Chrome, Edge) then you can get quick feedback by connecting to the micro:bit via "Connect" and flashing code after each change with "Flash". After the first flash the process is much quicker.

I'll leave this issue open to track the simulator feature request.

pddring, 2022-01-25:
There's a really interesting article on the Arm community page about research into building a simulator: https://community.arm.com/education-hub/b/rob-leeman/posts/building-a-micro-bit-simulator

I cobbled one together when the micro:bit first came out so that students could quickly test their code and large classes of students could share a small number of devices: e.g. https://create.withcode.uk/python/GH6

This simulator uses Skulpt (https://skulpt.org/) to attempt to transpile python into javascript so that it can run in the browser with the microbit module coded as a module that you can import. This means it's not a perfect replica of the micro:python interpreter but it's usually close enough for students not to notice most of the time. Source code is here: https://github.com/pddring/createwithcode but it was put together quickly rather than elegantly!