microbit-foundation/python-editor-v3 #450: Make “my program” a translatable string?

microbit-matt-hillsdon, 2022-01-20:
One for discussion. It's currently stored along with the initial program which isn't localised.

microbit-matt-hillsdon, 2022-01-21:
I think I didn't translate this as it wasn't translated in V2, but I think it's worth discussion:

https://python.microbit.org/v/2?lang=fr

microbit-matt-hillsdon, 2022-02-02:
We should localise "my program" and the comment at the top of the script.

If we can let's do it for user testing.

microbit-matt-hillsdon, 2022-02-02:
Let's discuss the comment here (and in a new file) further. Is there any value in either of them?

microbit-matt-hillsdon, 2022-02-03:
Short term: change to "Untitled project", make translatable.
Ensure that changing language before user edit changes text but not after.
Do this for user testing.

We don't think the comment at the start has much value. We'd like to encourage edits and reference Explore.
We should make it translatable.
@microbit-mark to suggest text for discussion.

microbit-mark, 2022-02-03:
Some permutations in https://paper.dropbox.com/doc/Python-script-intro--BbPcjaFoaBrHPGGzuz97CepcAg-sOSEZvjTdGZRytYWq330V

I would suggest something that is explicit and related to the most common user action of editing the code/script and flashing it to the device

[image]

I have tried some versions where we reference the Explore menu, but I wonder if doing so might make it really confusing if you take the script out of context. Eg if I open it in a text editor and it's talking about an 'Explore' menu

[image]

microbit-matt-hillsdon, 2022-02-03:
We should consider some kind of non-code notice/nudge too, if the main point is to communicate the above rather than to demonstrate a code comment.

microbit-matt-hillsdon, 2022-02-07:
Moved the issue of the default script to #566 along with @microbit-mark's suggestions.

Closing this as we've addressed the project name.

closed 2022-02-07 by microbit-matt-hillsdon