microbit-foundation/python-editor-v3 #522: [Toolkit] Enable copy/paste from CM embeds

microbit-matt-hillsdon, 2022-02-02:
The CodeMirror embeds doesn't support copy/paste. 

This is unrelated to the pop-up approach and seems to be due to CM being read-only (see CodeMirror view). If we upgrade CodeMirror we may have some new options to address this - see https://github.com/codemirror/codemirror.next/issues/368 and https://discuss.codemirror.net/t/codemirror-6-readonly-view/2333

This will need to wait until after user testing.

microbit-matt-hillsdon, 2022-10-05:
Worth noting we allow you to copy the whole snippet via Ctrl-C or the button when it's selected but this is the separate issue that you can't select the code from within the snippet (for larger examples you may just want a part of it).

Tagging @microbit-robert as we're looking at upgrading CodeMirror.

microbit-robert, 2022-10-06:
The real issue is the conflicting desire to make the drag handle as big as possible vs allowing the code to be selectable. We can configure the CodeMirror embeds to use `EditorState.readOnly.of(true)` instead of `EditorView.editable.of(false)`. This allows them to be focusable so they can be entered using keyboard navigation. The user can then select the text they want to copy. However, we override CTRL/CMD - C to copy the whole snippet, even in this scenario, so it's not much help without further changes.

microbit-matt-hillsdon, 2022-10-06:
Fair enough, in that case I think we should just close this and wait for feedback. Closing.

closed 2022-10-06 by microbit-matt-hillsdon