microbit-foundation/python-editor-v3 #525: Double space inserts a period on Mac, confusing users

microbit-matt-hillsdon, 2022-02-02:
This is a Mac system option but very confusing in a coding context.

Can we avoid it?

@microbit-katie's original report:

> Is there a way to add an in-line comment without needing to manually delete the period that auto populates? I tried it twice thinking it would sense that I didn’t want the period and stop auto-populating it (like Word formatting), but in the end had to manually delete the period after typing out my hash.

microbit-robert, 2022-02-02:
Previously raised here: https://github.com/microbit-foundation/python-editor-next/issues/71

Raised on the CodeMirror repo here: https://github.com/codemirror/CodeMirror/issues/6401

microbit-matt-hillsdon, 2022-10-06:
There's a slightly scary workaround suggested on the CM issue above. No real fix is possible.