microbit-foundation/python-editor-v3 #535: Clicking same search result twice has no effect

microbit-robert, 2022-02-03:
On clicking a search result, navigation to the clicked topic entry is activated. Clicking the same search again (without clicking any other result, or navigating away from the open topic) has no effect. This is problematic if the user has scrolled through the topic, and wishes to return the searched entry using the results.

microbit-robert, 2022-02-04:
Re-running the search and clicking on the previously clicked result **does** produce the correct navigation behaviour. E.g., search for `if`, click the `Nested if` result. Edit the search to backspace the f and retype it so a new result list is rendered. Click the `Nested if` result again and navigation works correctly.

microbit-matt-hillsdon, 2022-02-04:
> Re-running the search and clicking on the previously clicked result **does** produce the correct navigation behaviour. E.g., search for `if`, click the `Nested if` result. Edit the search to [...]

I think I know why. The results have a navigation entry associated with them which has { anchor: id } or similar. It is the identity of that object that causes navigation to (re)occur. So we should create the object at the point of navigation from the required info, not use one from the search results. Caveat: I didn't look at the code!

PR #552 (2022-02-04): Fix search results navigation
Clicking same entry twice in a row (or more) now results in the correct navigation behaviour.

Closes #535.
commit 38bff3cf (2022-02-04): Fix search results navigation (#552)

Clicking same entry twice in a row (or more) now results in the correct navigation behaviour.

Closes #535.
closed 2022-02-04 by microbit-matt-hillsdon