microbit-foundation/python-editor-v3 #594: Pyright logging has changed to log arrays in alpha

microbit-matt-hillsdon, 2022-02-14:
```
(2) ['[LS]', 'Searching for source files']
instrument.ts:125 (2) ['[LS]', 'No source files found.']
instrument.ts:125 (2) ['[LS]', 'Background analysis(Pyright-background-1) root directory: /']
instrument.ts:125 (2) ['[LS]', 'Background analysis(Pyright-background-1) started']
```

We're not clear why this has changed.

microbit-robert, 2022-02-21:
This was a bug introduced in sentry 6.17.7 and fixed in 6.17.8 - see issue https://github.com/getsentry/sentry-javascript/issues/4558.

This is now no longer a problem in the alpha and the console is logging strings again.

In our theme project, we use Sentry@6.6.0, but apparently this is not respected. See the output in this issue where @sentry/minimal is referenced as 6.16.1 - https://github.com/microbit-foundation/python-editor-next-microbit/issues/8. What is going on here?

microbit-matt-hillsdon, 2022-02-23:
I think it's just semver combined with the way we install the theme package.
Because we npm install it, we don't benefit from the lock file tracking the versions of its dependencies.
The only dependency is Sentry.
I've pinned it in the theme project and will upgrade to that version.

closed 2022-02-23 by microbit-matt-hillsdon