microbit-foundation/python-editor-v3 #606: Delay on hover makes you miss error context

microbit-mark, 2022-02-18:
Not sure if this is by design, but adding this for discussion as it came up in a user interview

When you whoever over an error, it seems to take a long time for the error message to appear.

[image]

If you don't dwell for long enough, you can easily miss the fact that there is some extra context available to help you fix the issue.

microbit-matt-hillsdon, 2022-02-18:
This will be a default from CodeMirror I suspect. I'll see if it's one we can configure.
It does seem slower to appear than equivalent functionality in VS code, though it's difficult to time.
I also remember seeing users have time to try other interactions (e.g. right click) so I agree it would be good to reduce.

Some short delay is desirable to avoid a mess moving the mouse across the code when there are many errors.

microbit-matt-hillsdon, 2022-02-18:
There's undocumented support at a low level: https://github.com/codemirror/tooltip/blob/main/src/tooltip.ts#L594 
but it's not used/accessible from the lint code: https://github.com/codemirror/lint/blob/main/src/lint.ts#L86

So I think this would need to be a CM feature request, either for configurability or a change to [the default of 600ms](https://github.com/codemirror/tooltip/blob/main/src/tooltip.ts#L468).

microbit-matt-hillsdon, 2022-02-18:
I think the VS Code equivalent is 300ms:

[image]

microbit-matt-hillsdon, 2022-02-18:
I asked on the CM forum: https://discuss.codemirror.net/t/hover-delay-for-lint-errors/4025

microbit-matt-hillsdon, 2022-02-18:
Default has been reduced to 300ms and made semi-configurable (at the hover/tooltip level not the lint level).
I think 300ms is OK based on my testing with VS Code, but if we really do need to customise we'd need to make another request or fork the lint module. I suggest we do the latter temporarily to prove the benefit.

https://github.com/codemirror/tooltip/commit/b14ea7f54ab0968dedb35558960ec78f00a52c17

microbit-matt-hillsdon, 2022-02-18:
Need a release of [@codemirror/tooltip](https://www.npmjs.com/package/@codemirror/tooltip) > 0.19.14 to upgrade.

microbit-robert, 2022-02-21:
The new default of 300ms will take effect when #563 is merged.

PR #563 (2022-02-22): Update Codemirror
Upgrade CodeMirror to latest version.

Closes #559 
Closes #606 - see https://discuss.codemirror.net/t/hover-delay-for-lint-errors/4025
commit 507b53b9 (2022-02-22): Update Codemirror (#563)

Closes #559
Closes #606 - see https://discuss.codemirror.net/t/hover-delay-for-lint-errors/4025
closed 2022-02-22 by microbit-matt-hillsdon