microbit-foundation/python-editor-v3 #663: Minimal "Ideas" section for beta

microbit-matt-hillsdon, 2022-04-07:
## Depends on

Content from @microbit-giles, itself blocked on the CMS structure for this.

## Description

Content model:

- A title
- A photo-like image
- V1/V2 support tag info similar to current toolkit content
- Rich text with code blocks, generally containing a single code example

UI will use a two level design similar to Reference but the top and second levels should make good use of the image.

Top level will show only image and title.

Second level will show all content with a breadcrumb nav back to the top level.

## Todo

- [ ] CMS structure
- [ ] Initial content (can be dev-created)
- [ ] UI
- [ ] Final content from @microbit-giles 
- [ ] Merge

## Conclusion

jaustin, 2022-04-08:
This won't be translated initially as the translation has started, but as there are a small number of strings here, perhaps we can include it in a second round?

microbit-matt-hillsdon, 2022-04-13:
Raised #672 for Ideas search.

I'll follow up separately on plans for translation as this has missed the window for the French for beta.

closed 2022-04-13 by microbit-matt-hillsdon