microbit-foundation/python-editor-v3 #695: Python editor is error

crouse12, 2022-04-21:
When I type serve -s build, the built python editor does not have any reference and ideas
[image]

[image]

[image]

microbit-matt-hillsdon, 2022-04-21:
Hi @crouse12, thanks for trying out the editor.

Please try http://localhost:3000 rather than http://127.0.0.1

The educational content in the sidebar is served from our content management system to allow our education team to maintain it. This requires some [CORS](https://developer.mozilla.org/en-US/docs/Web/HTTP/CORS) configuration on our end for each domain that needs to access the content. `localhost` is configured in this way but `127.0.0.1` is not.

If you just want to try the editor out, you can use https://python.microbit.org/v/beta
If you want to develop it locally, then run `npm start` for the development server that will rebuild as you make changes.
If you want to use `serve` then make sure you've run `npm install`  and `npm run build` first to create the `build` folder.

crouse12, 2022-04-22:
I shutdown my network and try npm start.
Python editor has an error when I try http://localhost:3000/
[image]
I want to have an offlinece python editor, because the network sometime shutdown in the school.
Microsoft give an offline version makecode for school.
Is is possible for python editor?
In addition, May I know how to create my reference if I need them like oled etc.?

microbit-matt-hillsdon, 2022-04-22:
> I shutdown my network and try npm start. Python editor has an error when I try http://localhost:3000/

This is for the same reason: the Reference/Ideas content is retrieved from the online content management system.

> I want to have an offlinece python editor, because the network sometime shutdown in the school. Microsoft give an offline version makecode for school. Is is possible for python editor?

Offline support is something we've discussed and are looking for user input to help prioritise.

MakeCode has service-worker-based offline support, so once you've visited MakeCode at https://makecode.microbit.org then you should still be able to load it even if you are offline. Would an equivalent setup meet your needs?

> In addition, May I know how to create my reference if I need them like oled etc.?

This isn't currently possible. Is "oled" a Python module you use in class? We're considering building an Extensions feature that might enable adding a Python module and its documentation in one step.