microbit-foundation/python-editor-v3 #707: Reference panel layout breaks when narrow with Display and Buttons categories

martinwork, 2022-04-22:
## Bug Description

Reference panel layout breaks when it is narrow with Display and Buttons categories.

## How To Reproduce

Steps to reproduce the behaviour:

1. Go to 'https://python.microbit.org/v/beta'
2. Make the Reference panel narrow
3. Select Display or Buttons
4. See error as in screenshot

## Expected behaviour

Works the same as other categories.

## Screenshots

[image]

## Environment

**Desktop (please complete the following information):**

- OS: [e.g. iOS] Windows 10
- Browser [e.g. chrome, safari] Chrome
- Version [e.g. 22] 100.0.4896.127

**Smartphone (please complete the following information):**

- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

## Additional context

Add any other context about the problem here.

jaustin, 2022-04-22:
Relates to #638 prehaps?

microbit-matt-hillsdon, 2022-04-22:
Fix merged. It should be on beta in < 10 mins. You can also try it via the link on #708.

PR #708 (2022-04-22): Fix for image responsiveness
This allows toolkit images to shrink below their set widths preventing the reference tab layout issue referenced below.

Closes #707
commit d828b59a (2022-04-22): Fix image widths, maintain responsiveness (#708)

This allows toolkit images to shrink below their set widths preventing the reference tab layout issue referenced below.

Closes #707
closed 2022-04-22 by microbit-matt-hillsdon