microbit-foundation/python-editor-v3 #718: Use the chance to explain flashing on the first flash dialogue

jaustin, 2022-04-29:
On this dialog
[image]

We could use this as a chance to tell people what flashing is.

Proposal for (a)
an image something like this
[image]

@microbit-lucy could you please propose text for the dialogue?

microbit-lucy, 2022-04-29:
I'm not sure it even needs a picture although great if we can add one easily (perhaps an arrow on the cable with the word code to show code going to the micro:bit and the word 'flashing' above). 

Your code is being sent ('flashed') to your micro:bit. 
Initial flash might take longer, subsequent flashes will be quicker.

We could even move to more friendly language for the 2nd sentence e.g. 

Your code is being sent ('flashed') to your micro:bit. 
This can take a while the first time, sorry, but it will be quicker after that.

jaustin, 2022-04-29:
I like the more friendly text, and let's continue with no image for now, and then add the image as we have one!