microbit-foundation/python-editor-v3 #764: Long filenames break the "Projects" layout

microbit-carlos, 2022-06-01:
## Bug Description

A long filename break the Projects layout and hides the file options.

## How To Reproduce

Steps to reproduce the behavior:

1. Go to 'Projects'
2. Create a new file with a very long name
4. See error

## Expected behavior

Layout to somehow wrap or adapt the file name to fit in the given width.

## Screenshots

[image]

## Environment

**Desktop (please complete the following information):**

- OS: macOS Monterey
- Chome v103

## Additional context

NA

microbit-robert, 2022-06-07:
Duplicate of #135. Closing.

microbit-carlos, 2022-06-07:
Oh right, thanks for linking to the original issue! Although to be honest with that title and description is fairly unlikely to be found by users searching to see if this had been reported before (as I did), so I'll add this content over there to make it more "search friendly".