microbit-foundation/python-editor-v3 #774: Stubs missing for `micropython.asm_thumb` decorator

microbit-carlos, 2022-06-20:
## Bug Description

The `@micropython.asm_thumb` decorator and assembly "functions" are not recognised by the code checker.

## How To Reproduce

Add this code to the editor:

```python
import micropython
import machine

FICR_PART_CODE = 0x10000100

@micropython.asm_thumb
def reg_read(r0):
    ldr(r0, [r0, 0])

print("machine   nRF part -> {}".format(hex(machine.mem32[FICR_PART_CODE])))
print("asm_thumb nRF part -> {}".format(hex(reg_read(FICR_PART_CODE))))
```

## Expected behavior

No errors to be indicated in the code editor

## Screenshots

[image]

## Environment

N/A

## Additional context

Low priority, but something to look into at some point.
Might be tricky to have all the assembly words as available "functions", but not in the autocompletion (or at least only when used inside a decorated function).

microbit-carlos, 2022-06-24:
> Do you know if this is used by accessory vendors or others?

I'm not familiar with a specific vendors using it, but there is at least this libraries online that people have been using (not sure if it originated from Vitta Science, but I've definitely seen it around): 
- https://github.com/vittascience/microbit-libraries/blob/58f5c5e59534bb6239090005ab665ac2b3f5b4bb/dht11.py
- https://github.com/vittascience/microbit-libraries/blob/a0d96613d9c4ee5b333021e8cd1e2ed11eeab3cb/dht11_v2.py

And this GitHub search shows a few more hits, but I haven't really looked into it, so can't tell how popular these might be:
- https://github.com/search?l=Python&p=2&q=microbit+asm_thumb&type=Code

> As you say, hard to see how we'll model this in the stubs. Maybe a pretty low-level hack to Pyright that matches the decorator and behaves similarly to an import statement at the top of the function.

That could work!

microbit-matt-hillsdon, 2022-10-10:
This also failed badly in the sim, see https://github.com/microbit-foundation/micropython-microbit-v2-simulator/issues/81

microbit-robert, 2024-03-21:
The following workaround can be applied to the code referenced in this issue. Add this line to the top of the file to ignore errors present:
`# pyright: reportGeneralTypeIssues=false, reportUndefinedVariable=false`

A list of diagnostic rules can be found in the [Pyright repo](https://github.com/microsoft/pyright/blob/main/packages/pyright-internal/src/common/diagnosticRules.ts).

A more targeted approach could to be to add `# type: ignore` to the end of problematic lines.