microbit-foundation/python-editor-v3 #782: Scrolling in siderbar can get stuck

microbit-matt-hillsdon, 2022-06-24:
Dubious repro (@microbit-robert couldn't match it):

In Chrome on Mac with a mouse with a mouse wheel:

1. Open Loops.
2. Size sidebar so the Loops description wraps
3. Position mouse under the header on the RHS of the sidebar
4. Scroll up/down repeatedly until you can't (~10 times for me)

microbit-carlos, 2022-06-24:
I've had this issue before! I can reproduce now as well with this instruction, but it usually works again when moving the mouse a bit.

kmcnaught, 2022-09-21:
Scrolling always gets stuck when a code block is highlighted, regardless of wrapping. Code blocks get highlighted when hovered over, which can trigger unintentionally during a "scrolling while skim-reading" action and give the appearance of sporadic behaviour. 

Is this the same issue discussed here? Video demo: https://www.loom.com/share/25f1ff1e15434badbe4a5d6387dafe11