microbit-foundation/python-editor-v3 #809: Unused function error only triggered if function doesn't start with `_`

microbit-carlos, 2022-07-10:
## Bug Description

I'm not sure if the "unused function" error is silenced, but there is a different in behaviour between `def foo(): ...` and `def _foo():...`

## How To Reproduce

This screenshot shows the issue, the "this function is unused" appears with `_foo` only. And it also breaks the structure highlighting:
[image]

```python
def foo():
    if True:
        while True:
            pass

def _foo():
    a = 10
    b = 100
    c = 1000
    if True:
        while True:
            pass
```

## Expected behavior

Both functions to error, or neither.
Code structure highlighting to continue working.

## Screenshots

In "How to reproduce" secton

## Environment

N/A

## Additional context

N/A

microbit-matt-hillsdon, 2022-10-07:
I think this intentional behaviour by Pyright, because _foo is not considered part of the public interface of the module and not used within it. We could attempt to change this but I'm not sure I see the value. I suspect folks using underscores in this way may be doing so for this distinction.

The effect on code structure highlighting seems like it might be an independent bug affecting function bodies that start with an assignment. I've split that into https://github.com/microbit-foundation/python-editor-v3/issues/1022 and hope to look at it soon.

microbit-matt-hillsdon, 2022-10-10:
Closing this as the _foo vs foo behaviour is as per Pyright's design which doesn't worth trying to deviate from here. The highlighting issues are on #1022 now.

closed 2022-10-10 by microbit-matt-hillsdon