microbit-foundation/python-editor-v3 #850: [Search] Cannot find "microbit.accelerometer" and similar module docs by typing accelerometer

microbit-matt-hillsdon, 2022-07-28:
You cannot find "microbit.accelerometer" and similar module docs by typing "accelerometer" in Search.

It works if you type the full module name of "microbit.accelerometer" but no-one is likely to do this.

We should reconsider what (or how) we index here to improve matching microbit.* module names.

microbit-carlos, 2024-04-11:
If the technical solution turns out to be too complex, a workaround could be to mention the module name in the description. for example, SPI shows up because of that (but uart, i2c, etc don't).
[image]

microbit-carlos, 2024-04-11:
Works a treat, thanks Rob!