microbit-foundation/python-editor-v3 #912: while True: radio.on() locks up sim

microbit-matt-hillsdon, 2022-08-26:
```python
while True:
    radio.on()
```

Appears to lock up the sim.

We should get a change to break out but we don't.

microbit-matt-hillsdon, 2022-08-26:
This doesn't seem to be an issue when using the simulator demo so perhaps we're just doing much more work on each simulator state change. Perhaps we should RAF rate-limit the simulator state updates.

microbit-matt-hillsdon, 2022-08-26:
It gets better with useRafState (stoppable again) but you can see re-rendering play out for some time after you click stop. Probably fixable with some re-render optimisation.

microbit-robert, 2022-10-11:
This is still currently an issue for me. The longer you let the while loop run, the worse the effect is as everything waits to catch up. The same thing happens with `radio.send()`. Forgetting a call to `sleep()` is probably going to be common error as well.

We only disable the radio on `radio.on()` in order to reconfigure the radio in `drv_radio.c`. Can we check whether the configuration has changed and only disabled/re-enable if it has? This won't help in the case of `radio.send()` so some form of rate-limiting will likely be required.