microbit-foundation/python-editor-v3 #973: Inconsistent tab order / initial focus

kmcnaught, 2022-09-20:
## Bug Description

I have been reviewing the tab order for keyboard/screen-reader users, and it appears that sometimes focus starts in the top left and follows a logical order top-to-bottom and left-to-right and other times it jumps straight into the simulator pane. Once underway the tab order is consistent, it's just like we've short-circuited it by focussing the simulator at an inopportune moment. 

## How To Reproduce

Sometimes I can reproduce this consistently by using a screen reader, or not. But I imagine this is purely coincidentally, since the page should not know whether a screen reader is being used. And it doesn't always occur consistently.

Steps to reproduce the behavior:

1. Load https://python.microbit.org/v/beta
2. Hit F5 to refresh to get consistent starting state
3. When page is loaded, press tab a few times and observe focussed elements

On several occasions I have seen the correct behaviour occurring when either Narrator or NVDA are running, and the incorrect behaviour occurring when they are not. As I started writing this issue I was observing the bug on the live editor, but right now I can only trigger it on a local version (2fd2bfd). 

## Expected behaviour
Tab order should start from top left like in screenshot below

## Screenshots
### Correct tab order
Note that there may be a separate discussion to be had about the best tab order, depending on whether the left hand pane is collapsed or expanded, but this certainly seems to be the intended behaviour.

[image]

### Incorrect tab order

[image]

## Environment

- OS: Windows 10
- Browser: Chrome (105.0.5195.127) and Firefox (104.0.2) 
- Version: Editor 0.1.0-dev.3084
MicroPython (micro:bit V1) 1.1.0-beta.1
MicroPython (micro:bit V2) 2.1.0-beta.3

microbit-matt-hillsdon, 2022-09-20:
It is likely that this issue was fixed by https://github.com/microbit-foundation/python-editor-v3/pull/900 merged as ac0a9ec963090f3e236ea50e2b88651fd2502c84 on September 8th vs your local commit from end of August.

kmcnaught, 2022-09-21:
@microbit-robert good spot that a soft/hard refresh may give different behaviour with NVDA - presumably it maintains some state on a soft refresh. The issues you describe are separate to the very specific behaviour I described above (but am struggling to replicate now, so could have been fixed by ac0a9ec). I'll open a separate issue to discuss the overall tab order concerns.

microbit-robert, 2022-09-21:
Reading around this, NVDA by design remembers where the user's last position when returning to a previously visited web page. This isn't working quite as expected with the Python Editor.

microbit-robert, 2022-09-21:
Somewhat more inexplicably, running the Editor locally yields different behaviour to the beta. Locally, I always start focussed in the simulator.

kmcnaught, 2022-09-21:
> Somewhat more inexplicably, running the Editor locally yields different behaviour to the beta. Locally, I always start focussed in the simulator.

Does this occur with and without NVDA running?

microbit-robert, 2022-09-21:
> > Somewhat more inexplicably, running the Editor locally yields different behaviour to the beta. Locally, I always start focussed in the simulator.
> 
> Does this occur with and without NVDA running?

This is without NVDA, but I suspect introduced by https://github.com/microbit-foundation/python-editor-v3/commit/ac0a9ec963090f3e236ea50e2b88651fd2502c84.

kmcnaught, 2022-09-21:
> 

I suspect if you run NVDA you will "fix" it, could you please test?

microbit-robert, 2022-09-21:
#982 has fixed all issues relating to focus and tab order with and without NVDA when the project is run locally (not inside an iFrame). When using the beta deployment (inside an iFrame), NVDA is causing the focus to switch to different areas of the Editor.

microbit-matt-hillsdon, 2022-09-21:
All being well with some other tasks, we'll remove the iframe for the Python Editor launch. However, we still plan to embed the app in the micro:bit classroom so it's worth us exploring it further in that context.

microbit-matt-hillsdon, 2022-09-23:
> All being well with some other tasks, we'll remove the iframe for the Python Editor launch. However, we still plan to embed the app in micro:bit classroom so it's worth us exploring it further in that context.

This issue should now be fixed for the beta deployment.
The micro:bit classroom scenario is affected by this issue (as it embeds the editor in an iframe) and we've raised an internal ticket for this.

closed 2022-09-23 by microbit-matt-hillsdon