microbit-foundation/python-editor-v3 #977: Review tab order for accessibility

kmcnaught, 2022-09-21:
Let's review this together on a call, but I'll highlight some 

## Bug Description

A clear and concise description of what the bug is.

## How To Reproduce

Steps to reproduce the behavior:

1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

## Expected behavior

A clear and concise description of what you expected to happen.

## Screenshots

If applicable, add screenshots to help explain your problem.

## Environment

**Desktop (please complete the following information):**

- OS: [e.g. iOS]
- Browser [e.g. chrome, safari]
- Version [e.g. 22]

**Smartphone (please complete the following information):**

- Device: [e.g. iPhone6]
- OS: [e.g. iOS8.1]
- Browser [e.g. stock browser, safari]
- Version [e.g. 22]

## Additional context

Add any other context about the problem here.

kmcnaught, 2022-09-21:
Overall, when all the sidebars are open, the tab order is fairly sensible, but I think there are some improvements to be made. Some areas for discussion...

## Navigation from a closed docs sidebar
If you're loading to a small size, the left hand docs sidebar is closed, and the Ref/Ideas/API/Project are skipped from the tab order. This means you can't tab into these areas without expanding the sidebar first - and the screen reader description doesn't signpost what the contents are. 
[image]

If instead you collapse the sidebar manually, then you get a much nicer tab order, which I think would be reasonable to copy.  
[image]

## Navigation to editor from an open docs sidebar
If the left hand sidebar is open (reference pane open by default), it takes >40 tabs to get to the editor, as you are taken via all the topic buttons. If you are using a screen reader and know all the shortcuts, you can shortcircuit by navigating by region, but it would be preferable to have a more streamlined / nested navigation order
[image]

## Navigation from revisited webpage
See discussion in 973 - Reading around this, NVDA by design remembers where the user's last position when returning to a previously visited web page. This isn't working quite as expected with the Python Editor.
_Originally posted by @microbit-robert in https://github.com/microbit-foundation/python-editor-v3/issues/973#issuecomment-1253422274_

See also 973

microbit-robert, 2022-09-21:
#982 has fixed the "Navigation from a closed docs sidebar" part of this issue. When using the beta deployment (inside an iFrame), NVDA is causing the focus to switch to different areas of the Editor - this does not occur when running the project locally (not in an iFrame).