microbit-foundation/python-editor-v3 #984: Button presses are sometimes holds

kmcnaught, 2022-09-21:
## Bug Description

In the simulator you can press button A or B. By default this is supposed to be a "press and release" unless you expand to access the "hold" toggle switches. 

However, some methods of clicking the buttons leave them incorrectly stuck down. So far I've found touch input, and one particular type or simulated mouse click. But I imagine it's a more general issue to do with the timings/specifics of mouse up/down/press events.

## How To Reproduce

Simplest way to reproduce is with a touch screen device:
1. Run the sample code in the simulator
2. Tap button A or B with your finger
  - expected behaviour: button is pressed and then released
  - observed behaviour: button is held down
3. Tap again and button is released.
4. Tap again and button is held down
5. Now click the held-down button with your mouse
  - The button now works in reverse, briefly being released while you click it and then being pressed again

The same scenario can be reproduced using [Optikey](https://github.com/OptiKey/OptiKey/) to simulate mouse presses with eye control.

## Environment

Windows 10/Chrome/Surface Pro touchscreen

microbit-matt-hillsdon, 2022-09-21:
This reproduces for me on iPad too. It doesn't affect the buttons on the board itself but it does affect the buttons in the Buttons section of the simulator controls.

microbit-robert, 2022-09-23:
We should be able to close this once the Editor is using the next release of the simulator and that use with Optikey has been checked.

microbit-matt-hillsdon, 2022-09-25:
The editor is now using the latest simulator since #994. I've not tested with Optikey so leaving this open for now.

microbit-robert, 2022-10-05:
Testing using Optikey, there is still an outstanding problem where clicking a button in the simulator module control effectively toggles the hold mode for the button. The simulator board works as expected. Will look into this shortly.

kmcnaught, 2022-10-06:
fwiw the simulated click from Optikey won't involve a [press ... wait ... release] action, but will be a single instantaneous "left click" action, which sometimes can cause issues if software is expecting a lag.

closed 2022-10-12 by microbit-robert

microbit-matt-hillsdon, 2022-10-14:
Now live on https://python.microbit.org in [v3.0.5](https://github.com/microbit-foundation/python-editor-v3/releases/tag/v3.0.5)

PR #1017 (2022-10-12): Fix for simulated button clicks for simulator button/pin inputs
Closes #984.