microbit-foundation/python-editor-v3 #1022: [Code structure highlighting] Function bodies not highlighted when first statement is an assignment

microbit-matt-hillsdon, 2022-10-07:
Split out from https://github.com/microbit-foundation/python-editor-v3/issues/809.

[image]

e, f, and g are incorrectly highlighted. It would be interesting to see what CodeMirror's parse tree is in this case.

This seems likely to arise in practice as initialising locals is common at the start of a function.

microbit-matt-hillsdon, 2022-10-07:
```python
def a():
    pass

def b():
    print("Hello")

def c():
    print("Hello")

def d(x):
    x = 1

def e(x):
    x += 1

def f():
    x, y = 1, 2

def g():
    a = 1
    print()
```

microbit-robert, 2022-10-10:
This occurs whenever a variable is unused in a function. Assignment can happen first, as long as the variable assigned is referenced later. Similarly, if the function takes an argument, and that argument is unused in the function, highlighting also breaks. Essentially, the `var_name is unused` hint/info diagnostic is interfering with the highlighting.

microbit-matt-hillsdon, 2022-11-10:
We intentionally hide code structure highlighting that overlaps with unused code diagnostics. Perhaps "overlaps" was the wrong call. We need to revisit and set up some scenarios to see how they look. I think we had things like content after `while True` in mind rather than unused locals.

microbit-robert, 2022-11-18:
It looks like we want to hide code structure highlighting for unreachable code, rather than unused code. Pyright bundles both unused and unreachable code into the same tag, DiagnosticTag.Unnecessary. If we could extend DiagnosticTag to include something like "Unreachable", then there is very little to do on the Editor side.