microbit-foundation/python-editor-v3 #1027: Duplicated Neopixel 'show'

DaveAtKitronik, 2022-10-11:
The neopixel api has show() and write() functions: 
[image]

They appear to do the same thing, but one is for V2 only. This seems unnecessary duplication, with the potential to cause a slightly mystifying failure if a user only has a V1 micro:bit: 
[image]

The autocomplete lists all the things: 
[image]

I haven't managed to find any 'code' implementing the new write function in the micropython repo (https://github.com/bbcmicrobit/micropython) so can't see why there is a V2 version.

Would suggest removing the duplicate write(), and leaving the show() as this has been the API forever.

DaveAtKitronik, 2022-10-12:
The 'same' functionality isn't the issue - although duplication is not great. 
The fact the new one doesn't work for older microbits will cause issues in the classroom - especially where there are mixes of V1 and V2 microbits being used.

I hadn't seen it before yesterday, but that's probably because I don't expect APIs to change, and we have to make as much as possible work on V1 and V2, so wouldn't be looking for V2 only API changes.

jaustin, 2022-10-24:
We've just confirmed with Damien that, indeed, the V2 `write()` method is included for compatibility with upstream MicroPython - we expect people to use `show()` in all cases on micro:bit. I don't think it's worth removing `write()` because it's useful for people using code shared across other MicroPython boards (on a V2), but when we add 'Reference' documentation for Neopixel, we'll only use V1/V2 compatible APIs for the common features.