microbit-foundation/python-editor-v3 #1028: Auto indent doesn't work as expected

DaveAtKitronik, 2022-10-11:
It is possible to write valid python code with any number of white space indents. The style guide recommends 4 spaces, and that is the default indent for the new editor. So far so good.
However if I write code that has (for instance) 3 spaces in it is valid:
[image]
And my text appears to be on the green highlight.
When I press return for a new line however instead of following the line above indent as expected I am presented with 4 space indents.:
[image]

And then a confusing error where the new line (which is on the mandated 4 spaces) is the one in error:
[image]

If 4 spaces are going to be mandated then the editor should never allow any other spacing, or it should follow the convention of all other editors I have used and auto indent to the value of the line above.

microbit-matt-hillsdon, 2022-10-12:
Students struggle with indentation so we've tried to provide a strong steer that 4 space indent is the correct approach vs the configurability you find in a more general purpose editor. I can see how this could feel like a frustrating middle ground though. There are stronger suggestions such as #755 that we've considered. I think this might be one for us to collect feedback on and then revisit.

DaveAtKitronik, 2022-10-12:
I'm not suggesting that configurable indent is necessarily required. 
I personally use 2 spaces most of the time, but that's a long term style for me based on years of typing in various languages... 

The issue I am trying to highlight is the editor allows indents of not 4 spaces, and the parser for 'good code' is also happy with it, but then the editor doesn't respect the user choice of less than 4 (or more than 4) as is the case in pretty much every editor, where blocks appear under the previous line indent.
The editor simply gives a 4 space indent on the next line no matter what. 
This is compounded for the user by the error message being created by the parser which is expecting 3 spaces (in this case) highlighting that the 4 space (auto applied) indent is in error.  
The highlighting in https://github.com/microbit-foundation/python-editor-v3/issues/755 would also show the error on the 4 space line. 

In this case it is also not helped by the 3 spaces being indicated as green as the letter does not overhang the highlight - so it looks at first glance to be correct.