microbit-foundation/python-editor-v3 #1082: SyntaxError: constant must be an integer

Thesnake66six, 2022-12-11:
Although code runs fine on the actual microbit, it throws this error in the simulator.

'''

microbit-matt-hillsdon, 2022-12-12:
Thanks for the report.

This seems to be a difference between MicroPython for the micro:bit V1 board version (where it works) and MicroPython for the micro:bit V2 where it does not. The simulator is based on MicroPython for micro:bit V2.

I'll get some input on why this difference exists and follow up here.

microbit-matt-hillsdon, 2022-12-12:
For V1 const isn't doing anything, it's simply returning its argument. This is unlikely to change at this point due to code size and implementation effort constraints. If you're free to change this code then you can simply remove the const call. 

For V2 we're using MicroPython 1.18.0 where using const is optimising the integer constants and intentionally failing for non-integer values.

Coincidentally, upstream MicroPython 1.19.1 [now has support](https://github.com/micropython/micropython/commit/079f3e5e5bc03f8e7af34a5e6e8db8bf2c570ba7) for more types of constant. We'll plan an upgrade in future which will align the behaviour (though V1 will still lack the optimisation).