microbit-foundation/python-editor-v3 #1084: Bluetooth support for micro:bit v2

nevergone, 2022-12-19:
## Description

Please add bluetooth support for Python Editor v3 with micro:bit v2.x. Or something solution for wireless micro:bit and external device (PC, mobile phone) connection.

## Todo

- [ ] Resolve radio <-> bluetooth conflict (remove conflicted extension)

microbit-matt-hillsdon, 2022-12-20:
Hi @nevergone, thanks for the report. This would first need to be supported in MicroPython for micro:bit V2. There's some discussion of this on https://github.com/microbit-foundation/micropython-microbit-v2/issues/51

nevergone, 2022-12-21:
Thanks! Ok, I understand.
For now, is it not possible to create a hex file with the same knowledge as apple from the micro:bit python editor? 
I would like to port external extensions to python.