microbit-foundation/python-editor-v3 #1085: Clicking through from microbit.org projects sometimes shows default code

microbit-matt-hillsdon, 2022-12-31:
A couple of times when I've clicked "Open in Python" on microbit.org on this project I've ended up with the default code:

https://microbit.org/projects/make-it-code-it/times-tables-tester/?editor=python

When I've gone to reproduce it it has worked correctly. Raising here as it's much more likely to be a bug in the editor than the website as I think the link would be identical in each case. This is the only project I've used recently so the issue could be more widespread.

microbit-robert, 2023-01-03:
I always get the default code for that project. We don't consider it a valid migration unless `v.meta?.name === "string"`, and the code section for that project is missing a name in the CMS.

microbit-matt-hillsdon, 2023-01-03:
> I always get the default code for that project. We don't consider it a valid migration unless `v.meta?.name === "string"`, and the code section for that project is missing a name in the CMS.

Missing typeof?

Perhaps I tested with a different project when it worked.

 Let's raise an internal website ticket and find out how widespread this is.

microbit-robert, 2023-01-03:
> Missing typeof?
>
> Perhaps I tested with a different project when it worked.
>
> Let's raise an internal website ticket and find out how widespread this is.

It has typeof, my failure to copy/paste properly: `typeof v.meta?.name === "string"`. Sounds good.

microbit-matt-hillsdon, 2023-01-03:
I raised a web ticket for this, we'll fix it on that end. I'm not opposed to relaxing the requirement in the Python Editor in future, but in this context it's useful for us to provide a name that includes the name of the Make it: code it, not just the name of the code embed (which is optional).

closed 2023-01-11 by microbit-robert