microbit-foundation/python-editor-v3 #1102: EOF issue using input

microbit-matt-hillsdon, 2023-02-11:
1. Using the program below, run it once and answer the first question
2. Press reset on the simulator
3. Notice that there's an immediate EOFError scrolling

After this restart doesn't work and stop seems to stop but doesn't show the play button again.

I tried a minimal program using `input()` and couldn't immediately reproduce it there.

```
# Imports go at the top
from microbit import *
import random
import music

# Code in a 'while True:' loop repeats forever
print("Welcome to my amazing times table game!")
for question in range(0, 10):
    a = random.randint(3, 12)
    b = random.randint(3, 12)
    while True:
        text = input("%s ÷ %s = " % (a * b, b))
        try:
            guess = int(text)
            break
        except:
            print("That wasn't a number!")
    if a == guess:
        audio.play(Sound.YAWN)
        print("You are awesome")
    else:
        music.play(music.BADDY)
        print("Loser :P")
    print()
    sleep(1000)
    
print("All done!")
```

microbit-matt-hillsdon, 2023-02-11:
Minimal reproduction:

```
input("Name? ") # reset when prompted
```

microbit-matt-hillsdon, 2023-02-11:
This also reproduced in the simulator demo page so I've rasised https://github.com/microbit-foundation/micropython-microbit-v2-simulator/issues/100 for the fix. I'll leave this open to track.

martinwork, 2023-03-29:
Reported in support ticket https://support.microbit.org/helpdesk/tickets/61614 (private) 

With the code below running in the simulator, after clicking the simulator Stop button or changing the code, it cannot be restarted.

I can usually get back control by sending Ctrl+D.
```
while True:
    name = input('What is your name? ')
    print('Hello', name)
```
[image]

microbit-matt-hillsdon, 2023-04-05:
This is now fixed and can be seen in the beta Python editor (which uses the simulator staging environment).

I've responded to the support ticket above.

I'll update the live simulator early next week if there's no issue with beta.

closed 2023-04-05 by microbit-matt-hillsdon