microbit-foundation/python-editor-v3 #1127: Simulator crash

martinwork, 2023-09-15:
Arising from https://support.microbit.org/helpdesk/tickets/66683 (private)

## Bug Description

Program modifying a global array inside too many (3) nested functions causes simulator to crash and appear to just hang until page is refreshed.

The browser F12 console reports and Uncaught RuntimeError.

Program runs OK in micro:bit.

Here's a simplified version of the program that triggers the same error.

It does not crash if start() calls work() directly, or if the contents of start() are unpacked to be the top level code, or if dis is not an array.

```
from microbit import *

dis = []

def work():
    global dis
    dis = []
    display.set_pixel( 0, 0, 0 if display.get_pixel(0, 0) else 9)
    
def loop():
    work()

def start():
    while True:
        loop()

start()
```

## How To Reproduce

Run the abobe program in the simulator

## Expected behavior

Report error in the program or in simulator and give helpful message to recover.

## Screenshots

[image]

## Environment

**Desktop (please complete the following information):**

- OS: [e.g. iOS] Windows
- Browser [e.g. chrome, safari] Chrome
- Version [e.g. 22] 117

microbit-matt-hillsdon, 2023-09-19:
Interesting, thanks Martin.

I've confirmed it reproduces on the [simulator demo page](https://python-simulator.usermbit.org/v/0.1/demo.html).

microbit-matt-hillsdon, 2023-09-19:
Simpler. Repros without the counting. Counts to 1981 in the version below, but less if you allocate more/bigger lists in c(). Also fails if you allocate a bytearray() and likely others. Works if you reduce the level of nesting.

```
def a():
    i = 0
    while True:
        print(i)
        i += 1
        b()

def b():
    c()

def c():
    []

a()
```

microbit-matt-hillsdon, 2023-09-19:
Raised a [simulator issue](https://github.com/microbit-foundation/micropython-microbit-v2-simulator/issues/105). Leaving this open to track (likely it will be fixed with a simulator upgrade alone).

microbit-matt-hillsdon, 2023-09-29:
The simulator fix is now live on https://python.microbit.org. Closing.

closed 2023-09-29 by microbit-matt-hillsdon

martinwork, 2023-09-29:
Thanks for this @microbit-matt-hillsdon, and for adding a response on the support ticket!