microbit-foundation/python-editor-v3 #1133: simulator/data log: estimate time until log is full

jaustin, 2023-10-09:
One of the things that's quite hard with a micro:bit and data logging is working out in advance how soon you will fill up the data log.

Once logging has started, given we have a fixed log size, it would be great to give an indication of how long it would take the log to fill up at the current rate of logging.

Without this, a user has to
a) work out how large the log is (120kB)
b) work out how much data they're logging each log/per second
c) calculate how long that would take to fill up

Or, just test it, but if the time to fill the log is, say, 20 minutes (or even 1 day), that could be really annoying.

jaustin, 2023-10-09:
See also https://github.com/microsoft/pxt-microbit/issues/5368

microbit-matt-hillsdon, 2023-10-10:
We have this UI for a full log at the moment so perhaps we could use the same place to show the rate of fill.

[image]

"Estimated time until log is full: 30 mins" feels quite long but I struggle to imagine something shorter that's clear. A small progress bar with that as a hover interaction? If you've filled the log very quickly it would be useful to get the stat for a full log too.