microbit-foundation/python-editor-v3 #1148: Publish Sanity schema

felixerdy, 2024-03-07:
## Depends on

Other tickets:

- N/A

Third parties:

- N/A

## Description
I'm interested in forking the editor and enhancing the toolkit with some custom functions/code. Upon reviewing your repository, I couldn't locate the underlying Sanity CMS schema used to request content for the toolkit, ideas, and API. Would it be possible for you to share the schema model?

microbit-matt-hillsdon, 2024-03-19:
Hi @felixerdy. Sorry for the delayed response.

The schema is currently part of a larger schema but we'll look at how best to extract the relevant part. It looks like we might be able to extract it as a sanity plugin which we could then open source. We’ll take a look and get back to you soon.

The schema covers the content model for the Reference and Ideas tab content as well as a mapping from Reference content to API entries. The Reference and Ideas content itself is not open source.

The API content is managed differently: it's generated at runtime from the Python type stubs that we provide to our fork of Pyright. These stubs are available [in this repo](https://github.com/microbit-foundation/micropython-microbit-stubs). There's a script there to generate the typeshed.{lang}.json file that we use to initialize Pyright.

We're also considering spending some time on an alternative way to source Reference and Ideas tab content. For example, a version based on markdown in the git repository. Is that something that would be useful to you? We'd be unlikely to switch to such an approach ourselves (there are benefits to using Sanity for collaboration with our education team) so it would be a more open-source-friendly alternative.

felixerdy, 2024-03-21:
Thank you for your answer @microbit-matt-hillsdon! Publishing the schema as a sanity plugin would be great of course, but the markdown approach would also be a great alternative. If I find some time I'll try to work on a first draft with markdown files (or something similar) that we could discuss on

microbit-matt-hillsdon, 2024-03-26:
@felixerdy we've created an initial cut of a Sanity plugin with Python Editor schemas here: https://github.com/microbit-foundation/sanity-plugin-python-editor-v3/

We hope the README will be enough to add it to a blank Sanity studio given some experience with Sanity itself. Feedback very welcome.

We made some supporting changes in the Python Editor:
- https://github.com/microbit-foundation/python-editor-v3/pull/1157
- https://github.com/microbit-foundation/python-editor-v3/pull/1158

It's also worth noting that we recently updated the tooling from Create React App to Vite/Vitest as one of these changes use Vite-flavoured environment variable configuration (this change on main and live on our beta deployment but not yet released).

I'll spend some time later in the week planning out what a markdown alternative could look like. I'll create a new issue and tag you there. There are some features of the current documentation (particularly the dropdowns with alternative content) that don't have an obvious mapping to markdown so I'll try to capture those. Once done I'd be interested in your input on what features are important to you for your content. We've also considered using markdown documentation for extension module documentation so it would be good to consider that usecase.

microbit-matt-hillsdon, 2024-03-27:
Markdown tracked by #1160.

microbit-matt-hillsdon, 2024-04-02:
Added some docs on #1159. Calling this done but please feel free to add feedback here or open a new issue for any problems.

closed 2024-04-02 by microbit-matt-hillsdon