microbit-foundation/python-editor-v3 #1189: Can not connect to microbit on Linux

hinxx, 2024-07-09:
## Bug Description

Can not connect to microbit.

## How To Reproduce

Steps to reproduce the behavior:

1. https://python.microbit.org/v/3/project
2. Open new project (demo code is fine)
3. Connect microbit
4. Send the code to microbit
5. Try to pair the microbit (seen in the list)
6. Get error: WebUSB error : Please reconnect your micro:bit and try again.

## Expected behavior

Be able to download code to microbit.. FWIW, I can connect to the same device with https://makecode.microbit.org/#editor. I'm only trying one editor at a time.

## Environment

- OS: Linux Mint 21
- Browser chome (brave)
- Version  1.67.116 Chromium: 126.0.6478.71 (Official Build) (64-bit)

## screenshots

[image]
[image]

## Additional context

My microbit:
```
# DAPLink Firmware - see https://daplink.io
Build ID: v0257-gc782a5ba (gcc)
Unique ID: 99063602000528207dbe0e669db29888000000006e052820
HIC ID: 6e052820
Auto Reset: 1
Automation allowed: 0
Overflow detection: 0
Incompatible image detection: 1
Page erasing: 0
Daplink Mode: Interface
Interface Version: 0257
Bootloader Version: 0257
Git SHA: c782a5ba907377658bc28aa8d132a0fa44543687
Local Mods: 0
USB Interfaces: MSD, CDC, HID, WebUSB
Bootloader CRC: 0x725bea7d
Interface CRC: 0xe561f1de
Remount count: 0
URL: https://microbit.org/device/?id=9906&v=0257
```

Developer console in browser shows this:

```
board-serial-info.ts:20 USB serial number unexpected length: 16
webusb.ts:134 An error occurred whilst attempting to use WebUSB.
webusb.ts:134 Details of the error can be found below, and may be useful when trying to replicate and debug the error.
webusb.ts:134 Error: Could not recognise the Board ID b
    at new cl (board-id.ts:16:13)
    at cl.parse (board-id.ts:56:12)
    at oA.parse (board-serial-info.ts:25:40)
    at get boardSerialInfo (dap-wrapper.ts:63:28)
    at uSe.reconnectAsync (dap-wrapper.ts:89:29)
    at async mSe.connectInternal (webusb.ts:386:5)
    at async webusb.ts:170:13
    at async mSe.withEnrichedErrors (webusb.ts:318:14)
    at async GLe.connectInternal (project-actions.tsx:192:13)
    at async GLe.flash (project-actions.tsx:508:25)
    at async SendButton.tsx:61:15
webusb.ts:134 Device status NOT_CONNECTED
webusb.ts:301 Disconnection complete
```

hinxx, 2024-07-09:
Given `board-serial-info.ts:20 USB serial number unexpected length: 16`:
Here is what my lsusb shows for microbit:

[lsusb.txt](https://github.com/user-attachments/files/16147994/lsusb.txt)

microbit-matt-hillsdon, 2024-07-10:
Thanks for reporting. 

We're parsing the serial to determine whether to flash MicroPython for a V1 or V2 micro:bit. The serial in lsusb output parses correctly.

The issue reproduces in Brave cross platform.

It seems that Brave has an anti-fingerprinting feature that randomises the USB device's serial number. We'll give some thought to what we can do in this scenario. In the meantime, if you're happy to do so, in my testing it works successfully with the fingerprinting setting changed for the site:

[image]

microbit-matt-hillsdon, 2024-07-10:
I think this is how MakeCode is making the same decision without requiring the serial: https://github.com/microsoft/pxt-microbit/blob/bd4bda19f33ba30ea61e3620c8fa069b8306d07e/editor/flash.ts#L326

hinxx, 2024-07-10:
> In the meantime, if you're happy to do so, in my testing it works successfully with the fingerprinting setting changed for the site:

Yeah, that works.

Thanks for looking into this!

microbit-carlos, 2024-08-30:
> I think this is how MakeCode is making the same decision without requiring the serial: https://github.com/microsoft/pxt-microbit/blob/bd4bda19f33ba30ea61e3620c8fa069b8306d07e/editor/flash.ts#L326

I can confirm that DAPLink can provide the "unique id" via DAP custom vender command 0x80:
- https://github.com/ARMmbed/DAPLink/blob/v0258/source/daplink/cmsis-dap/DAP.h#L78
- https://github.com/ARMmbed/DAPLink/blob/v0258/source/daplink/cmsis-dap/daplink_vendor_commands.h#L27
- https://github.com/ARMmbed/DAPLink/blob/v0258/source/daplink/cmsis-dap/DAP_vendor.c#L71

microbit-matt-hillsdon, 2024-08-30:
I think ideally we'd fix this in the work-in-progress connection lib and then get the fix in the Python Editor that way.