microbit-foundation/python-editor-v3 #1194: String missing from crowdin

martinwork, 2024-11-05:
Arising from https://crowdin.com/editor/microbitorg/6422/en-de?view=comfortable&filter=basic&value=3#328740

String "Heap memory can still be allocated, and garbage collection can still be initiated manually using gc.collect." is missing from crowdin

https://github.com/search?q=org%3Amicrobit-foundation+%22Heap+memory+can+still+be+allocated%22&type=code

https://crowdin.com/editor/microbitorg/all?view=multilingual&filter=basic&value=3#q=Heap+memory+can+still+be+allocated

Fabi135, 2024-11-05:
Mostly all of the other strings when clicking "Show more" under any of the API explanations are also missing from Crowdin. 
Here is a list of some that I found: 

- "returns: The number of bytes allocated. 
This function is MicroPython extension." 
under gc > `mem_alloc()`. 
  - https://github.com/search?q=org%3Amicrobit-foundation+%22The+number+of+bytes+allocated.%22&type=code
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=The+number+of+bytes+allocated.

- "returns: The number of bytes free.
This function is MicroPython extension." 
under gc > `mem_free()`. 
  - https://github.com/search?q=org%3Amicrobit-foundation+%22The+number+of+bytes+allocated.%22&type=code 
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=The+number+of+bytes+allocated.

- "This function is a MicroPython extension. CPython has a similar function - set_threshold(), but due to different GC implementations, its signature and semantics are different." 
under gc > `threshold()`. 
  - https://github.com/search?q=org%3Amicrobit-foundation+%22implementations%2C+its+signature+and+semantics+are+different.%22&type=code
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=implementations%2C+its+signature+and+semantics+are+different.

- "Normally, a collection is triggered only when a new allocation cannot be satisfied, i.e. on an out-of-memory (OOM) condition. If this function is called, in addition to OOM, a collection will be triggered each time after amount bytes have been allocated (in total, since the previous time such an amount of bytes have been allocated). amount is usually specified as less than the full heap size, with the intention to trigger a collection earlier than when the heap becomes exhausted, and in the hope that an early collection will prevent excessive memory fragmentation. This is a heuristic measure, the effect of which will vary from application to application, as well as the optimal value of the amount parameter." 
under gc > `threshold(amount)`. 
  - https://github.com/search?q=org%3Amicrobit-foundation+%22Normally%2C+a+collection+is+triggered+only+when+a+new+allocation%22&type=code
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=Normally%2C+a+collection+is+triggered+only+when+a+new+allocation

- "Ideally this function should be called a single time, before any data is logged, to configure the data table header once.
If a log file already exists when the program starts, or if this function is called multiple times, it will check the labels already defined in the log file. If this function call contains any new labels not already present, it will generate a new header row with the additional columns.
By default the first column contains a timestamp for each row. The time unit can be selected via the timestamp argument."
under log > `set_labels(*labels, timestamp=SECONDS)`
  - https://github.com/search?q=org%3Amicrobit-foundation+%22Ideally+this+function+should+be+called+a+single+time%22&type=code&p=1
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=Ideally+this+function+should+be+called+a+single+time

- "Each call to this function adds a row to the log.
New labels not previously specified via the set_labels function, or by a previous call to this function, will trigger a new header entry to be added to the log with the extra labels.
Labels previously specified and not present in a call to this function will be skipped with an empty value in the log row."
under log > `add(data_dictionary)`
  - https://github.com/search?q=org%3Amicrobit-foundation+%22Each+call+to+this+function+adds+a+row+to+the+log.%22&type=code
  - https://crowdin.com/editor/microbitorg/all?view=multilingual&languages=de&filter=basic&value=3#q=Each+call+to+this+function+adds+a+row+to+the+log.

Fabi135, 2024-11-05:
This is the same as the following issue: #1069