microbit-foundation/python-editor-v3 #1199: String "Code copied, use paste to insert it." missing from Crowdin.

Fabi135, 2024-12-14:
The string "Code copied, use paste to insert it." is missing from Crowdin. 

https://crowdin.com/editor/microbitorg/all/en-de?view=side-by-side&filter=basic&value=0#q=Code%20copied,%20use%20paste%20to%20insert%20it.

https://github.com/search?q=repo%3Amicrobit-foundation%2Fpython-editor-v3%20%22Code%20copied%2C%20use%20paste%20to%20insert%20it.%22&type=code

Attached is a screenshot of the string: 
[image]

microbit-matt-hillsdon, 2024-12-16:
Thanks for reporting @Fabi135. We've fixed this in the code and will update it in Crowdin later today.

closed 2024-12-16 by microbit-matt-hillsdon

Fabi135, 2024-12-16:
@microbit-matt-hillsdon  👍 Thank you