microbit-foundation/python-editor-v3 #1209: Show serial does not immediately show either print() output or error traceback

martinwork, 2025-03-07:
## Bug Description

Arising from https://support.microbit.org/helpdesk/tickets/87116 (private)

After pasting in code or loading a hex with a print() statement, then sending it to micro:bit, and clicking Show serial, the serial panel remains blank. If there's a code error, the traceback doesn't show either.

Switching to another browser tab and back fixes it.

Example: [serial.zip](https://github.com/user-attachments/files/19132703/serial.zip)
```
from microbit import *
count = 0
while True:
    display.show(Image.HEART)
    print( count)
    count += 1
    sleep(1000)
    display.show(Image.HEART_SMALL)
    sleep(1000)
```

## How To Reproduce

Steps to reproduce the behavior:

1. Go to https://python.microbit.org/
2. Paste in the example
3. Send to micro:bit
4. Click Show serial
5. See error

## Expected behavior
print() output and traceback appears in the serial window

## Environment

**Desktop (please complete the following information):**

- OS: [e.g. iOS] Windows 10
- Browser [e.g. chrome, safari] Chrome
- Version [e.g. 22] 133

microbit-matt-hillsdon, 2025-03-12:
@microbit-grace please can you reproduce this and explore whether this might be a regression due to the connection library update?

microbit-grace, 2025-03-12:
The issue seems to have existed pre-introduction of the connection library. I can recreate the issue on v3.0.27 (before introduction of connection library).

microbit-grace, 2025-04-01:
Carlos found that doing the following also seems to fix the serial output:
1. Disconnect using the “Disconnect” option in the “send to micro:bit button”
2. Connect by using the “Connect” option in the “send to micro:bit button”

This narrows down why switching tab might fix the serial output.

closed 2025-04-02 by microbit-matt-hillsdon